kantlivelong/OctoPrint-PSUControl-RPiGPIO #3: The GPIO channel has not been set up as an OUTPUT

DerDingsCS, 2021-04-16:
As decribed here [https://github.com/kantlivelong/OctoPrint-PSUControl/issues/180#issuecomment-821117122](https://github.com/kantlivelong/OctoPrint-PSUControl/issues/180#issuecomment-821117122) I cant turn the Pin on or off.
2021-04-16 13:38:48,701 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 13:38:48,705 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-16 13:39:49,020 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-16 13:39:49,024 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT

kantlivelong, 2021-04-16:
Please provide full logs as requested

kantlivelong, 2021-04-16:
@DerDingsCS Without having logs but taking a guess I think I might see why you're getting that error. Try setting the desired config then restarting OctoPrint.

tonesto7, 2021-04-16:
I'm seeing the same thing here is my log
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/files/6325914/octoprint.log)

DerDingsCS, 2021-04-16:
@kantlivelong CONFIRMED! Restart after setting everything up fixes the issue.
Thank you for your help and the plugins of cause ;-)

kantlivelong, 2021-04-16:
Awesome. I think I know what needs to be done from here. Simple issue with reconfiguring after changes.

DerDingsCS, 2021-04-16:
I'm sorry for not giving the logs and thankfull you was able to help without them. I'm a father of a 3 month old and my freetime is limited.

kantlivelong, 2021-04-18:
Should be fixed in 1.0.1

closed 2021-04-18 by kantlivelong