kantlivelong/OctoPrint-PSUControl-RPiGPIO #4: Can't control GPIO after v1.0.0

tonesto7, 2021-04-16:
2021-04-16 09:37:33,708 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-16 09:37:33,709 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 17
2021-04-16 09:37:33,710 - octoprint.plugins.psucontrol - ERROR - [Errno 2] Opening GPIO chip: No such file or directory
2021-04-16 09:38:46,689 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-16 09:38:46,690 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 17
2021-04-16 09:38:46,691 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.
2021-04-16 09:39:13,485 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:39:13,487 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:39:26,208 - octoprint.plugins.ws281x_led_status - INFO - Turning lights on
2021-04-16 09:39:34,285 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:39:34,287 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:40:12,871 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-16 09:40:12,872 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 17
2021-04-16 09:40:12,873 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.
2021-04-16 09:40:20,501 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:40:20,502 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:40:22,990 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:40:22,992 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:40:25,487 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:40:25,488 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:45:31,610 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:45:31,613 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:46:00,940 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-16 09:46:00,941 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 17
2021-04-16 09:46:00,944 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.
2021-04-16 09:46:07,648 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:46:07,651 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-16 09:46:09,908 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 09:46:09,910 - octoprint.plugins.psucontrol - ERROR - 'switch'

kantlivelong, 2021-04-16:
The log provided does not reference this plugin.

tonesto7, 2021-04-16:
Umm... It literally says psucontrol in the prefix

kantlivelong, 2021-04-16:
This repo is psucontrol_rpigpio

kantlivelong, 2021-04-16:
https://github.com/kantlivelong/OctoPrint-PSUControl

tonesto7, 2021-04-16:
Ahh... I see why I got confused.  I didn't read... 
Sorry for wasting your time

closed 2021-04-16 by tonesto7