kantlivelong/OctoPrint-PSUControl-RPiGPIO #10: Lightning icon no longer working (on/off isn't working)

Zappline, 2021-04-18:
Using RPI.GPIO after setting it up as best as I'm able.
General: board
Switching: pin 3 (made sure its the correct one)
Sensing: 0

After hitting save the printer turns on. However the lightning bolt does not turn green or indicate the printer is on. Clicking it does nothing.

Went back in to settings and ticked "invert" and hit save. This turns the printer off, and the lightning is greyed out. However, no matter how many times I hit it. It won't turn green and the printer does not power up.

So the only way for me at this moment is to go into settings and tick or unstick the invert button to turn the printer on and off... Rather annoying.

It might be something easy I just can't figure out or there is a rather annoying issue here.

Being able to download and install an older version (that I know works) of the add-on would have been nice and would have saved my from a serious headache. But couldn't find that as an option either.

Printer: Artillery Sidewinder X1 (if that helps)
Firmware: Klipper
OctoPrint: 1.5.3

kantlivelong, 2021-04-18:
Fill out the required info for new tickets.

Zappline, 2021-04-18:
> Fill out the required info for new tickets.

I think this is the best I can do. I'm not very good at these things.

kantlivelong, 2021-04-18:
Sorry but I really can't and won't do anything without logs.

Zappline, 2021-04-18:
> Sorry but I really can't and won't do anything without logs.

Probably doing something wrong. But I can't generate any logs. But it's OK. Turns out over at the main page. A bunch of people have have the same issue. Guess I'll just keep my eyes open over there. Until then. I'll try to figure out how to install a pre 1.0.0 version

kantlivelong, 2021-04-18:
Alright

closed 2021-04-18 by kantlivelong