kantlivelong/OctoPrint-PSUControl-RPiGPIO #12: no plugin-settings visible

thirschbuechler, 2021-04-20:
#### What were you doing?

Installing the Plugin and trying to configure it

#### What did you expect to happen?

Find a configuration page under octoprint settings / plugins / OctoPrint-PSUControl-RPiGPIO

#### What happened instead?

A white tab-page was displayed with no settings (as in #7).

Deleting the default configuration of the newly installed plugin (OctoPrint-PSUControl-RPiGPIO) solved the issue. 

This issue was created as it reproduced that behaviour (no settings visible) and might affect new users

#### Version of OctoPrint-PSUControl

1.0.3

#### Version of OctoPrint-PSUControl-RPiGPIO

1.0.2

#### Operating System running OctoPrint

octoprint 1.5.3
octopi 0.17

#### Printer model & used firmware incl. version

Anycubic i3 mega

#### Attach or link DEBUG log files.

n/a

#### Wiring diagram

not relevant

kantlivelong, 2021-04-20:
>Attach or link DEBUG log files.
>n/a

Really need logs to do anything.

thirschbuechler, 2021-04-20:
sorry. Does `~/.octoprint/logs $ cat octoprint.log | egrep -i psu`
help?

https://pastebin.com/1UqbetFR

kantlivelong, 2021-04-20:
Unfortunately nothing of use there. Do you have a backup of your settings from a date prior to upgrading?

As a side note you may be getting `octoprint.plugins.psucontrol - ERROR - [Errno 2] Opening GPIO chip: No such file or directory` due to https://github.com/kantlivelong/OctoPrint-PSUControl/issues/200. 

I'll hold this ticket open for a bit so if anyone wants to chime in with logs see https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

adamoutler, 2021-04-21:
I had this issue prior to 2-updates ago. I think this issue is resolved.  However, I still don't see a power button though. So I'm using both the PSU control and PSU.rpi.  I'll perform the update from today and file a new issue if required.  Just wanted to say this issue is resolved for me with no known changes on my side.

kantlivelong, 2021-04-21:
> I had this issue prior to 2-updates ago. I think this issue is resolved. However, I still don't see a power button though. So I'm using both the PSU control and PSU.rpi. I'll perform the update from [...]

A handful of bugs were addressed so it is entirely possible that it is now fixed. I just couldn't reproduce it.

I'll keep this open for a bit anyway.

PS: Give the core plugin a try again if you have time. You might not even need this sub-plugin.

kantlivelong, 2021-04-22:
> 
> 
> Plugin doesnt work. Same here, blank setting page. Cant control my printer anymore. Broken. So need to remove all the stuff to get printer running again. What a shit. Why not make 2 plugins and let the one running which was running fine.

It will stay that way unless you wanna pitch in some info.

BlutVampir, 2021-04-22:
At the highlighted pin is my Power Relay attached:
[url=https://ibb.co/1mzNwL0]image 1 of my pin[/url]
GPIO 2 or Board 3.

 [url=https://ibb.co/HYR4Gjj]BCM 2 not working at all (as I had GPIO 3 used before, I did test also 3 here, same effect, not working)[/url]
 [url=https://ibb.co/FJphzJq]board 3 not working[/url]

Config wiped completly and tried without plugin:
[url=https://ibb.co/vX52Z9g]GPIO 2 not working (as I had GPIO 3 used before, I did test also 3 here, same effect, not working)[/url] 

deinstalled everything and reinstalled version 0.1.11:
[url=https://ibb.co/Z6CjgQC]this setting is working.[/url]

What did I wrong?

kantlivelong, 2021-04-22:
> At the highlighted pin is my Power Relay attached:
> [url=https://ibb.co/1mzNwL0]image 1 of my pin[/url]
> GPIO 2 or Board 3.
>
> [url=https://ibb.co/HYR4Gjj]BCM 2 not working at all (as I had GPIO 3 used [...]

Open a new ticket and fill out all required info plus whatever extra you may think helps.

kantlivelong, 2021-04-25:
I was able to reproduce this. Fixed via 428033b2f8a1c573bf91105350d50e34c633e0cb. Will be part of 1.0.3.

For those affected simply restart OctoPrint after installing PSUControl-RPi.GPIO

commit 428033b2 (2021-04-25): Require restart on install - Fixes #12
commit 57d767bf (2021-04-25): Require restart on install - Fixes #12
closed 2021-04-25 by kantlivelong